markniu/PandaPi #1: Request for information (Thingiverse)

AJMartel, 2019-10-20:
I contacted you on Thingiverse.

Would you be able to give me the exact version of Marlin that you used?

This way I will be able to figure out exactly what changes and additions you made to make Marlin work on the RaspberryPi. Together we will be able to port your project over to Marlin v2.x.

By communicating through Github, we can track the progress of the project.
You will remain in control of this fork and I will become a contributor to your project, any changes/additions would be passed to you through the Pull requests giving you full control of any changes.

daimein2007, 2019-10-20:
https://github.com/markniu/PandaPi

markniu, 2019-10-21:
Hi @AJMartel  the exact version of Marlin is the latest released version 1.1.9 last year.

AJMartel, 2019-10-22:
Looks like you originally where using https://github.com/hackaday3D/marlin_stm32
which was based off of Marlin 1.1 - 2018 Aug 2 Patch level
Some of the fixes do not line up, I will find the starting point.

AJMartel, 2019-10-23:
stm32 is included in the Marlin v2 branch porting it over should be easier than remaining on the v1.1 branch, I am still trying to figure out all the relevant changes required.

markniu, 2019-10-24:
yes, Marlin2.0 has the HAL for different MCU