markniu/PandaPi #2: M851 not available

LeonKnook, 2020-01-04:
I have not received the board but I'm preparing the PI. I have managed to burn the image, wifi and all that and started Octoprint with no problem at all. What I noticed is that the command M851 is not available and I like that command a lot to adjust the printing height after the probing with my conductive probe. I could not found anywhere the right settings in Marlin to activate M851. Any suggestion or alternative method to adjust the printing height with auto bed tilt height correction after probing?

markniu, 2020-01-05:
the M851 needs that one of the bed probe been enabled.you can just enable the "BLTOUCH" in the Configuration.h even you have not bltouch but conductive probe，for that z probe pin and z endstop pin is the same.

LeonKnook, 2020-01-05:
[image] 
Good advice that solved the problem. Thank you Mark!

[image]