markniu/PandaPi #5: Files

furkanbr, 2020-01-17:
Hi , this is a great project and i want to work on this , can you share platformio.ini file for us?

markniu, 2020-01-18:
it is not compiled on platformio, it is compiled and run on raspberry pi Linux.  just run make command in the Linux

furkanbr, 2020-01-20:
How is it compiling ? is it happening automaticaly when i copy source code to root directory ?

and can i use my Rpi 3 a+ or it should be minimum 3b?

markniu, 2020-01-20:
both Pi3b and Pi3b+  are OK , here is the edit and compile steps 
 https://github.com/markniu/PandaPi/wiki/How-to-Edit-Marlin-code

furkanbr, 2020-01-25:
Can you tell me how did yo change serial port comminication in marlin firmware? im trying to do somethink at my arm stm32f4 but i didt figure that .

markniu, 2020-01-26:
You can  compare the it with original marlin code, e.g. marlinseral.cpp

Moorviper, 2021-04-04:
I think this can be closed.