markniu/PandaPi #6: Error when clicking run

Klinenator, 2020-01-18:
Im getting this error after click verify then run

cp: cannot stat '/media/usb/pandapi//pi_marlin': No such file or directory

Klinenator, 2020-01-18:
I believe the issue is that when it goes to make the file pi_marlin there is an error which is causing the error above

markniu, 2020-01-18:
after click verify it will compile the source and make the pi_marlin. did you see some error log after you click the verify?

Klinenator, 2020-01-19:
stderr:temperature.h: In static member function ‘static void Temperature::updatePID()’:
temperature.h:581:17: warning: comparison of integer expressions of different signedness: ‘int’ and ‘size_t’ {aka ‘unsigned int’} [-Wsign-compare]
    for(int i=0;i<strlen(tmp_data);i++)
                ~^~~~~~~~~~~~~~~~~
stderr:temperature.h:597:17: warning: comparison of integer expressions of different signedness: ‘int’ and ‘size_t’ {aka ‘unsigned int’} [-Wsign-compare]
    for(int i=0;i<strlen(tmp_data);i++)
                ~^~~~~~~~~~~~~~~~~
stderr:temperature.h:613:17: warning: comparison of integer expressions of different signedness: ‘int’ and ‘size_t’ {aka ‘unsigned int’} [-Wsign-compare]
    for(int i=0;i<strlen(tmp_data);i++)
                ~^~~~~~~~~~~~~~~~~
stderr:temperature.h:577:9: warning: unused variable ‘e’ [-Wunused-variable]
    char e=0;
         ^
stderr:Marlin_main.cpp: In function ‘void setup()’:
Marlin_main.cpp:14693:35: error: too many arguments to function ‘void step_motor_init(void (*)())’
    step_motor_init(stepper.isr,4*8); //mark
                                   ^
stderr:In file included from Marlin_main.cpp:288:
/home/pi/src/CMSIS/driver_api.h:74:6: note: declared here
 void step_motor_init(void(*step_isr)(void));
      ^~~~~~~~~~~~~~~
Marlin_main.cpp:14695:34: error: too many arguments to function ‘void step_motor_init(void (*)())’
    step_motor_init(stepper.isr,12); //mark
                                  ^
In file included from Marlin_main.cpp:288:
/home/pi/src/CMSIS/driver_api.h:74:6: note: declared here
 void step_motor_init(void(*step_isr)(void));
      ^~~~~~~~~~~~~~~
Marlin_main.cpp:14697:33: error: too many arguments to function ‘void step_motor_init(void (*)())’
    step_motor_init(stepper.isr,8); //mark
                                 ^
In file included from Marlin_main.cpp:288:
/home/pi/src/CMSIS/driver_api.h:74:6: note: declared here
 void step_motor_init(void(*step_isr)(void));
      ^~~~~~~~~~~~~~~
stderr:Marlin_main.cpp: In function ‘int test(int, char**)’:
Marlin_main.cpp:15038:41: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
    GetCurrentPath(buf,"Config_panda.ini");
                                         ^
stderr:Marlin_main.cpp:15044:68: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
    iCatAge = GetIniKeyInt("CAT","age","/media/usb/Config_panda.ini");
                                                                    ^
Marlin_main.cpp:15044:68: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
Marlin_main.cpp:15044:68: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
Marlin_main.cpp:15045:58: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
    char *a = GetIniKeyString("CAT4","name",g_szConfigPath);
                                                          ^
Marlin_main.cpp:15045:58: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
stderr:Marlin_main.cpp:15042:9: warning: unused variable ‘szCatName’ [-Wunused-variable]
    char szCatName[32];
         ^~~~~~~~~
Marlin_main.cpp:15056:6: warning: unused variable ‘cn’ [-Wunused-variable]
  int cn=0;
      ^~
stderr:Marlin_main.cpp: At global scope:
Marlin_main.cpp:15001:16: warning: ‘i2cMode’ defined but not used [-Wunused-variable]
 static uint8_t i2cMode = 0;
                ^~~~~~~
stderr:Marlin_main.cpp:14974:1: warning: ‘int GPIOWrite(int, int)’ defined but not used [-Wunused-function]
 GPIOWrite(int pin, int value)
 ^~~~~~~~~
Marlin_main.cpp:14949:1: warning: ‘int GPIORead(int)’ defined but not used [-Wunused-function]
 GPIORead(int pin)
 ^~~~~~~~
stderr:Marlin_main.cpp:14924:1: warning: ‘int GPIODirection(int, int)’ defined but not used [-Wunused-function]
 GPIODirection(int pin, int dir)
 ^~~~~~~~~~~~~
Marlin_main.cpp:14905:1: warning: ‘int GPIOUnexport(int)’ defined but not used [-Wunused-function]
 GPIOUnexport(int pin)
 ^~~~~~~~~~~~
Marlin_main.cpp:14885:1: warning: ‘int GPIOExport(int)’ defined but not used [-Wunused-function]
 GPIOExport(int pin)
 ^~~~~~~~~~
stderr:make: *** [Makefile:24: Marlin_main.o] Error 1
make: Leaving directory '/media/usb/pandapi'
stdclose:2

Klinenator, 2020-01-19:
Thats what i get when i try to make in that directory manually

markniu, 2020-01-20:
it seems that some of the file not update, "Marlin_main.cpp:14695:34: error: too many arguments to function ‘void step_motor_init(void ()())’",   check or download again the pandapi folder, which pi do you used pi3+or pi4? and which img file did you download?

Klinenator, 2020-01-20:
Pi4 and the pi4 img

markniu, 2020-01-20:
I will check it later today

markniu, 2020-01-20:
sorry for the out of update source for pi4, and now you can update the new github pandapi foler.

Klinenator, 2020-01-22:
Ok the update worked

Now i get this error when changing to TMC2130

 error: #error "Y_CS_PIN is required for TMC2130. Define Y_CS_PIN in Configuration_adv.h."
   #error "Y_CS_PIN is required for TMC2130. Define Y_CS_PIN in Configuration_adv.h."

markniu, 2020-01-22:
you can run as a4988 or tmc2208 model with TMC2130 first. niujl123 Email:niujl123@sina.com Signature is customized by Netease Mail Master On 01/22/2020 09:33, Klinenator wrote: Ok the update worked Now i get this error when changing to TMC2130 error: #error "Y_CS_PIN is required for TMC2130. Define Y_CS_PIN in Configuration_adv.h." #error "Y_CS_PIN is required for TMC2130. Define Y_CS_PIN in Configuration_adv.h."