markniu/PandaPi #7: PID autotune does not work

LeonKnook, 2020-02-19:
Today I updated everything possible in de marlin files to make PID autotune work.
It does not heat up at all.
Over here a zip file with my marlin version so far, maybe I overlooked something?

[pandapi.zip](https://github.com/markniu/PandaPi/files/4223976/pandapi.zip)

markniu, 2020-02-20:
hello, V1.0 board needs to update new MCU firmware,
https://github.com/markniu/PandaPi/tree/master/firmware
https://github.com/markniu/PandaPi/wiki/How-to-flash-MCU-firmware

LeonKnook, 2020-02-29:
Purchased the ST-Link USB module and uploaded the new firmware. I confirm that the PID autotune works now. In my case the temperature is unfortunately 5 degrees to low.

markniu, 2020-02-29:
did you update the latest firmware and marlin code that I had updated the yesterday? how about the default Value of PID?or try to ''M303 C10 S210  '' instead of "M303 S210"

LeonKnook, 2020-02-29:
I refreshed the 3 marlin files you uploaded yesterday and the latest firmware I already had. 
[image]
[image]
These are the results
[image]

LeonKnook, 2020-02-29:
And then I printed something
[image]
Still around 5 degrees to cold.

markniu, 2020-02-29:
it seems that the RPI  print out wrong PID ,I am doing test. set the PID default : 
M301 P25 I2.4 D66.9
M500

LeonKnook, 2020-03-01:
The PID default give bad results.
[image]

markniu, 2020-03-01:
sorry for that bad temperature. so I do more test and fixed it today,but need more test,if you have time you can try to update this new firmware and marlin . following is my test results. 
[image]   the PID value:  P25.00 I2.40 D66.90

LeonKnook, 2020-03-01:
temperature.h replaced and latest hex file geupload.
[image]

[image]

markniu, 2020-03-03:
the I0.002 is too low, try  I1, or increase P and D value.   when the autotune failed, try again.did it failed everytime?

markniu, 2020-03-14:
I find that the bed temperature on you board are not stable, because of the noise of the hardware or connect line,so I fixed it with the software,you can update the new firmware.
[image]