markniu/PandaPi #9: printer halted. kill() called

kine3000, 2020-03-13:
I get this error "printer halted.kill() called" everytime I want to home the axis. Theres no more information about the error written on the terminal.
Also I cant move any axis anyway even before this error occuring.
I've got an pandapi v2.0 with an raspberry pi 4b with octoprint running. Connection is fine. Temperatures are fine. Marlin code editing works fine. I use the original Marlin firmware version out of the "pandapi" folder with small corrections. Drivers are all A4988's and also defined this way in marlin. I have mechanical endstops and defined them correctly also checked them using M119.
I just dont get any driving voltage to the motors. I checked the correct wiring of the 4 pin connections as well. Vdd and Vmot on the drivers is correct.
I dont have any lcd display but that shouldnt be the problem.

What else can I check?

markniu, 2020-03-14:
you can check the En pin on the A4988, that should be low,and the motors can be locked.
also check the GPIO40 connector between PRI and pandapi is connected well.
test it with only one A4988 plugged in first. 
Be careful the driver is easily damaged.

kine3000, 2020-03-17:
I can disable/enable the motor now with the M17/M18 command. The En pin is working fine and in lock mode i have voltage on both spoils of the motor. But there is no moving with G1 command and there is still the problem of the error message coming up with homing. The driver voltage is set to 0.625V and i checked other ones as well so I guess that shouldnt be the Problem too.
It seems to me like a software problem but i cant think of any marlin configurations that would be responsible for this.
Maybe there are some other ideas how to come to a solution?
Thanks for support! Its great!

markniu, 2020-03-18:
yeah,if the motor is locked that means En is working,if can not move probably the step pin has no input signal. so the step pin needs to be checked, and can you move other motors with G1?

kine3000, 2020-03-30:
I'm having a more deeper problem in connection right now i think. After connecting successfully with octopi and sending M17/M18 to enable/disable the motors there is no change on the pi gpio17 (pin 11) at all, but the connection is still working like i get temperatures informations and can control the fan etc.
I also updated the pandapi image and the marlin 2.0 on the u disk without any problem.
Maybe somebody has any ideas cause i dont have anymore :D