markniu/PandaPi #12: Compile STM32F103C8 firmware from source

wmzart, 2020-04-14:
Hello,

Where can I find the source of the firmware for the STFM32F103C8 controller?

I would like to try to build the source myself.

Thanks!

Marc

wmzart, 2020-04-14:
User @keczejo already mentioned that the "... source code for stm is not public yet ...". [(See here)](https://github.com/markniu/PandaPi/issues/11#issuecomment-609736296). So when are these files going to be released?

markniu, 2020-04-15:
After selling 1,000 of this board, the firmware will be open source, :)

wmzart, 2020-04-15:
Cool! That would be fantastic.

wmzart, 2021-04-02:
Either you have sold at least 1000 of these boards or you changed your mind. Nevertheless a big thanks for providing the source!

https://github.com/markniu/PandaPi/tree/master/firmware/pandapi_stm32f103

closed 2021-04-02 by wmzart