markniu/PandaPi #13: [info] Started to draw vector graphics for documentation.

Moorviper, 2020-05-14:
[image]
Hi mark for you to know i am currently do some vector graphics of your board which you can use for documentation.
It is not finished yet ... but in the work.

markniu, 2020-05-15:
Great! Many others want this 3d file that I don't have.

Moorviper, 2020-05-15:
[image]
Unsure if i rotate all text in one direction. 
I don't draw the resistors but i have to check for the led's :D

markniu, 2020-06-11:
that's great.Thank you so much!  
about the fans, I just draw this [image](https://github.com/markniu/PandaPi/wiki/How-to-wire),hope that help! 
[image]

I learn the Altium Designer tool, and had draw this[ board 3D file](https://drive.google.com/drive/folders/1Imnye9MxCWCUlUC2pw_A6A5WoUFMV1xj?usp=sharing) this week.

closed 2020-06-11 by markniu