markniu/PandaPi #15: suggestion:  Differentiate 24v and non-24v fan sockets

ChrisWeiss, 2020-06-12:
I really like the support for multiple fan voltages. It might be helpful to use different style sockets for the 24v fan sockets than the non-24v fan sockets, when things are that close together, it can be easy to plug things in incorrectly. With fans, the user may not find out until the fan burns out after a few hours at the wrong voltage. 

Maybe change socket style or color between the two types?

Moorviper, 2020-06-13:
Only the 2nd row is adjustable.

markniu, 2020-06-13:
That's a good suggestion, I will change the color of non -24v fan sockets.

markniu, 2020-08-09:
[image]

closed 2020-08-09 by markniu