markniu/PandaPi #17: after update no connection from octoprint to Marlin possible

Moorviper, 2020-07-08:
When i update octoprint and pandapi i only get

`
onnecting to: /dev/tnt0`
`Changing monitoring state from "Offline" to "Opening serial port"`
`Connected to: Serial<id=0x6a6214f0, open=True>(port='/dev/tnt0', baudrate=115200, bytesize=8, parity='N', stopbits=1, `
`timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor`
`Changing monitoring state from "Opening serial port" to "Connecting"`
`Send: N0 M110 N0*125`
`Unexpected error while writing to serial port: SerialException: 'write failed: [Errno 22] Invalid argument' @ ``comm.py:_do_send_without_checksum:3490`
`Please see https://faq.octoprint.org/serialerror for possible reasons of this.`
`Connection closed, closing down monitor`
`Changing monitoring state from "Connecting" to "Offline (Error: SerialException: 'write failed: [Errno 22] Invalid argument' @` `comm.py:_do_send_without_checksum:3490)"`
`Closing down send loop
`
after compiling Marlin and a reboot

Moorviper, 2020-07-31:
It is fixed with the newest image.

closed 2020-07-31 by Moorviper

loempiaboer1000, 2021-03-11:
hello,
 i almost got the package running on my own image (due reasons im not able to use the image from pandapi itself). how did u fix this error?