markniu/PandaPi #21: Stepper motor makes hissing noise

chibisuke, 2020-08-10:
I today received my first PandaPi and I was attemping to test it, but I'm having the following problem now:

The Steper motor that is connected to the extruder port is making a really louse hissing noise. 
I Tried:
- swapping the driver (TMC2209)
- Swaping the stepper motor
- Swaping the extruder port on the pandaPI board (E0 to E1)

Since im after swapping pretty much everything out the problem can't be related to any hardare component (Except maybe the pandapi board itself, I don't know), I suspect a software issue.

Any hints on how to troubleshoot this further?

Moorviper, 2020-08-10:
its a wiring issue

chibisuke, 2020-08-11:
Thx for your reply. Unfortunately I'm pretty sure it got nothing to do with wireing, sind I switched around the cabels, steppers and drivers already.

The only thing that made a difference is changing the UART ID of the driver, so I would strongly guess that its a problem with the config that gets sent to the driver by marlin via uart.

markniu, 2020-08-11:
this is strange! you can set the drivers as standalone mode to see if it's uart problem,just put the MS1/MS2 jumpers on, and pull out the jumppers near the raspberry GPIO.

chibisuke, 2020-08-11:
In standalone mode the problem is gone, so it definitly got somthing to do with the UART config that is sent. 

Any more ideas on how to procede to get the UART running properly?

markniu, 2020-08-11:
does this noise happens only in extruder port?

chibisuke, 2020-08-11:
In the default setup it only happens on the extruder port. If I setup any other port (x,y - didn't test z) as having UART ID 3 on the driver, the problem moves there.

markniu, 2020-08-11:
did you changed some configure? I will check it later today or tomorrow.

Moorviper, 2020-08-11:
Normally when you have the hissing sound you have to much current or a bad wiring.
[image]

chibisuke, 2020-08-11:
@markniu only for testing to troubleshoot it, but reverted it completely. 

@Moorviper yea, but if it would be a wiring issue why would it only occure on a stepper connected to a driver that has UART ID 3 set and disapear if I switch the driver to standalone mode? that doesn't make sense at all.
To much current would propably heat up the stepper very quickly, but would still be an option, since as far as I understood the TMC2209 can control the current via uart. 
Even though I already tried to check this in the config files, and didn't find anything, there still might be something I'm overlooking there.

chibisuke, 2020-08-13:
Still no idea where this is comming from.

Its not appearing in standalone mode, but as soon as I set the UART jumper in the board the problem is there, if one of the stepper is configure to use UART ID 2 (both jumpers set). 
If I'm giving it another ID then 2 (like remove both jumpers, it works, but obviously the config is not what its surposed to be then. 

Any ideas?

chibisuke, 2020-09-29:
In the meanwhile I switched to an SKR 1.4, with the same stepper drivers, same cabels, same steppers and the same settings. There is works without any problems. So either the PandaPI board, has a serios problem somewhere or the hacked marlin in making trouble. 

anyway, since there is no support comming from the developers here, the purchase was just a huge waste of money to me, so anyone considder buying a pandapi board, I can only strongly recommend... don't do it.