markniu/PandaPi #22: Creality CR-20: No LCD & Endstop does not stop movement

sergiofenoll, 2020-08-13:
Hello!

I am using a PandaPi v2.5 with my Creality CR-20, followed the flashing and wiring instructions, but I do not get any output on the LCD (I think maybe I need to install U8glib, but I'm not sure how to do that since it's not explained in the Configuration.h file).

Also, while moving the extruder works, when the endstop is reached, it doesn't stop automatically.

Any ideas about what I'm doing wrong?

Moorviper, 2020-08-13:
Try:
[image]

[image]

markniu, 2020-08-14:
the U8glib source code has been contained in and will be auto compiled.

markniu, 2020-08-14:
Did it reach the endstop and not stop while homing ?