markniu/PandaPi #23: I been try to enable a dual z but i get this error at the time of compile the file.

EMDM25, 2020-08-31:
^
src/module/temperature.cpp:2682:39: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
src/module/temperature.cpp:2685:38: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
   parse_string(cmd_buf,"B:","",out,&k);
                                      ^
src/module/temperature.cpp:2685:38: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
stderr:src/module/temperature.cpp:2688:39: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
   parse_string(cmd_buf,"T1:","",out,&k);
                                       ^
src/module/temperature.cpp:2688:39: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
src/module/temperature.cpp:2707:39: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
    parse_string(cmd_buf,"h","T",out,&k);
                                       ^
stderr:src/module/temperature.cpp:2707:39: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
#arm-linux-gnueabihf-g++ -I./  -I./src/lcd/dogm  -I/home/pi/src/CMSIS -I/home/pi/src/utility -ITMC2208Stepper -I/home/pi/src/CMSIS-W -I./lib/TMCStepper-master/src/source/-I./lib/TMCStepper-master/src/ -I./lib/NeoPixel_LED/ -Wall -g0 -lm -O0 -c     -std=c++11 -D PANDAPI -Wall -cpp src/module/stepper.cpp Configuration.h Configuration_831_1639.h Configuration_831_212.h Configuration_adv.h Configuration_adv_831_1639.h Configuration_adv_831_212.h Version.h pins_PANDA_PI.h pins_PANDA_PI_831_1639.h pins_PANDA_PI_831_212.h -o src/module/stepper.o  
#arm-linux-gnueabihf-g++ -I./  -I./src/lcd/dogm  -I/home/pi/src/CMSIS -I/home/pi/src/utility -ITMC2208Stepper -I/home/pi/src/CMSIS-W -I./lib/TMCStepper-master/src/source/-I./lib/TMCStepper-master/src/ -I./lib/NeoPixel_LED/ -Wall -g0 -lm -O0 -c     -std=c++11 -D PANDAPI -Wall -cpp src/module/temperature.cpp Configuration.h Configuration_831_1639.h Configuration_831_212.h Configuration_adv.h Configuration_adv_831_1639.h Configuration_adv_831_212.h Version.h pins_PANDA_PI.h pins_PANDA_PI_831_1639.h pins_PANDA_PI_831_212.h -o src/module/temperature.o  
stderr:In file included from src/module/tool_change.cpp:30:
src/module/temperature.h: In static member function ‘static void Temperature::setTargetHotend(int16_t, uint8_t)’:
src/module/temperature.h:602:17: warning: comparison of integer expressions of different signedness: ‘int’ and ‘size_t’ {aka ‘unsigned int’} [-Wsign-compare]
    for(int i=0;i<strlen(tmp_data);i++)
                ~^~~~~~~~~~~~~~~~~
stderr:src/module/temperature.h: In static member function ‘static void Temperature::setTargetBed(int16_t)’:
src/module/temperature.h:676:17: warning: comparison of integer expressions of different signedness: ‘int’ and ‘size_t’ {aka ‘unsigned int’} [-Wsign-compare]
    for(int i=0;i<strlen(tmp_data);i++)
                ~^~~~~~~~~~~~~~~~~
stderr:In file included from src/module/tool_change.cpp:30:
src/module/temperature.h: In static member function ‘static void Temperature::updatePID()’:
src/module/temperature.h:794:17: warning: comparison of integer expressions of different signedness: ‘int’ and ‘size_t’ {aka ‘unsigned int’} [-Wsign-compare]
    for(int i=0;i<strlen(tmp_data);i++)
                ~^~~~~~~~~~~~~~~~~
stderr:src/module/temperature.h:810:17: warning: comparison of integer expressions of different signedness: ‘int’ and ‘size_t’ {aka ‘unsigned int’} [-Wsign-compare]
    for(int i=0;i<strlen(tmp_data);i++)
                ~^~~~~~~~~~~~~~~~~
stderr:src/module/temperature.h:826:17: warning: comparison of integer expressions of different signedness: ‘int’ and ‘size_t’ {aka ‘unsigned int’} [-Wsign-compare]
    for(int i=0;i<strlen(tmp_data);i++)
                ~^~~~~~~~~~~~~~~~~
stderr:src/module/temperature.h:790:9: warning: unused variable ‘e’ [-Wunused-variable]
    char e=0;
         ^
#arm-linux-gnueabihf-g++ -I./  -I./src/lcd/dogm  -I/home/pi/src/CMSIS -I/home/pi/src/utility -ITMC2208Stepper -I/home/pi/src/CMSIS-W -I./lib/TMCStepper-master/src/source/-I./lib/TMCStepper-master/src/ -I./lib/NeoPixel_LED/ -Wall -g0 -lm -O0 -c     -std=c++11 -D PANDAPI -Wall -cpp src/module/tool_change.cpp Configuration.h Configuration_831_1639.h Configuration_831_212.h Configuration_adv.h Configuration_adv_831_1639.h Configuration_adv_831_212.h Version.h pins_PANDA_PI.h pins_PANDA_PI_831_1639.h pins_PANDA_PI_831_212.h -o src/module/tool_change.o  
make: Leaving directory '/home/pi/PandaPI/Marlin2.x/pandapi'
compile failed !stdclose:2compile failed !

markniu, 2020-09-01:
Hi, this log is short and have not contain the error infor. but at the same time I received your another emails that have have the error infor. 
    "void beginSerial(uint32_t) = delete; // Your platform does not currently support Software Serialstderr:make: *** [Makefile:36: src/module/stepper/trinamic.o] Error 1"
    the pandapi use only one hardware uart TX pin for all the drivers.
so if you use TMC2209 for Z2, uncomment "#define Z2_HARDWARE_SERIAL customizedSerial1" in the pins_PANDA_PI.h instead of use "Z2_SERIAL_TX_PIN "if you still have problems, you can send me your changed configure files,I will check it.

EMDM25, 2020-09-01:
That was the problem Sorry I didn't know that I have to enable it there.

EMDM25, 2020-09-01:
But know when I try to enable ZStepperAlign I got this error.

arm-linux-gnueabihf-g++ -O0 -c     -std=c++11 -D PANDAPI  -I./lib  -I./lib/NeoPixel_LED  -I./lib/TMCStepper-master  -I./lib/TMCStepper-master/examples  -I./lib/TMCStepper-master/examples/Calibrate_spreadCycle  -I./lib/TMCStepper-master/examples/Live_tune  -I./lib/TMCStepper-master/examples/Simple  -I./lib/TMCStepper-master/examples/Software_SPI  -I./lib/TMCStepper-master/examples/StallGuard  -I./lib/TMCStepper-master/examples/StallGuard_TMC2209  -I./lib/TMCStepper-master/examples/StallGuard_TMC2660  -I./lib/TMCStepper-master/examples/TMC_AccelStepper  -I./lib/TMCStepper-master/extras  -I./lib/TMCStepper-master/src  -I./lib/TMCStepper-master/src/source  -I./src  -I./src/core  -I./src/feature  -I./src/feature/bedlevel  -I./src/feature/bedlevel/abl  -I./src/feature/bedlevel/mbl  -I./src/feature/bedlevel/ubl  -I./src/feature/dac  -I./src/feature/digipot  -I./src/feature/leds  -I./src/feature/mmu2  -I./src/feature/touch  -I./src/gcode  -I./src/gcode/bedlevel  -I./src/gcode/bedlevel/abl  -I./src/gcode/bedlevel/mbl  -I./src/gcode/bedlevel/ubl  -I./src/gcode/calibrate  -I./src/gcode/config  -I./src/gcode/control  -I./src/gcode/eeprom  -I./src/gcode/feature  -I./src/gcode/feature/advance  -I./src/gcode/feature/baricuda  -I./src/gcode/feature/camera  -I./src/gcode/feature/cancel  -I./src/gcode/feature/caselight  -I./src/gcode/feature/clean  -I./src/gcode/feature/controllerfan  -I./src/gcode/feature/digipot  -I./src/gcode/feature/filwidth  -I./src/gcode/feature/fwretract  -I./src/gcode/feature/i2c  -I./src/gcode/feature/L6470  -I./src/gcode/feature/leds  -I./src/gcode/feature/macro  -I./src/gcode/feature/mixing  -I./src/gcode/feature/pause  -I./src/gcode/feature/powerloss  -I./src/gcode/feature/power_monitor  -I./src/gcode/feature/prusa_MMU2  -I./src/gcode/feature/runout  -I./src/gcode/feature/trinamic  -I./src/gcode/geometry  -I./src/gcode/host  -I./src/gcode/lcd  -I./src/gcode/motion  -I./src/gcode/probe  -I./src/gcode/scara  -I./src/gcode/sd  -I./src/gcode/stats  -I./src/gcode/temp  -I./src/gcode/units  -I./src/HAL  -I./src/HAL/HAL_PANDAPI  -I./src/HAL/HAL_PANDAPI/fastio  -I./src/HAL/HAL_PANDAPI/inc  -I./src/HAL/LINUX  -I./src/HAL/LINUX/hardware  -I./src/HAL/LINUX/inc  -I./src/HAL/LINUX/include  -I./src/HAL/LPC1768  -I./src/HAL/LPC1768/inc  -I./src/HAL/LPC1768/include  -I./src/HAL/LPC1768/u8g  -I./src/HAL/LPC1768/win_usb_driver  -I./src/HAL/SAMD51  -I./src/HAL/SAMD51/inc  -I./src/HAL/shared  -I./src/HAL/shared/backtrace  -I./src/HAL/STM32  -I./src/HAL/STM32/inc  -I./src/HAL/STM32F1  -I./src/HAL/STM32F1/dogm  -I./src/HAL/STM32F1/inc  -I./src/HAL/STM32F1/maple_win_usb_driver  -I./src/HAL/STM32_F4_F7  -I./src/HAL/STM32_F4_F7/inc  -I./src/HAL/STM32_F4_F7/STM32F4  -I./src/HAL/STM32_F4_F7/STM32F7  -I./src/HAL/TEENSY31_32  -I./src/HAL/TEENSY31_32/inc  -I./src/HAL/TEENSY35_36  -I./src/HAL/TEENSY35_36/inc  -I./src/inc  -I./src/lcd  -I./src/lcd/dogm  -I./src/lcd/dogm/fontdata  -I./src/lcd/dwin  -I./src/lcd/extui  -I./src/lcd/extui/lib  -I./src/lcd/extui/lib/anycubic  -I./src/lcd/extui/lib/dgus  -I./src/lcd/extui/lib/dgus/fysetc  -I./src/lcd/extui/lib/dgus/hiprecy  -I./src/lcd/extui/lib/dgus/origin  -I./src/lcd/extui/lib/dgus/pandapi  -I./src/lcd/extui/lib/ftdi_eve_touch_ui  -I./src/lcd/extui/lib/ftdi_eve_touch_ui/archim2-flash  -I./src/lcd/extui/lib/ftdi_eve_touch_ui/ftdi_eve_lib  -I./src/lcd/extui/lib/ftdi_eve_touch_ui/ftdi_eve_lib/basic  -I./src/lcd/extui/lib/ftdi_eve_touch_ui/ftdi_eve_lib/extended  -I./src/lcd/extui/lib/ftdi_eve_touch_ui/ftdi_eve_lib/extended/unicode  -I./src/lcd/extui/lib/ftdi_eve_touch_ui/ftdi_eve_lib/extended/unicode/font_bitmaps  -I./src/lcd/extui/lib/ftdi_eve_touch_ui/ftdi_eve_lib/extras  -I./src/lcd/extui/lib/ftdi_eve_touch_ui/language  -I./src/lcd/extui/lib/ftdi_eve_touch_ui/screens  -I./src/lcd/extui/lib/ftdi_eve_touch_ui/theme  -I./src/lcd/extui/lib/mks_ui  -I./src/lcd/HD44780  -I./src/lcd/language  -I./src/lcd/menu  -I./src/lcd/menu/game  -I./src/lcd/utility  -I./src/libs  -I./src/libs/heatshrink  -I./src/libs/L64XX  -I./src/module  -I./src/module/stepper  -I./src/module/thermistor  -I./src/pins  -I./src/pins/esp32  -I./src/pins/linux  -I./src/pins/lpc1768  -I./src/pins/lpc1769  -I./src/pins/mega  -I./src/pins/rambo  -I./src/pins/ramps  -I./src/pins/sam  -I./src/pins/samd  -I./src/pins/sanguino  -I./src/pins/stm32f0  -I./src/pins/stm32f1  -I./src/pins/stm32f4  -I./src/pins/stm32f7  -I./src/pins/teensy2  -I./src/pins/teensy3  -I./src/sd  -I./src/sd/usb_flashdrive  -I./src/sd/usb_flashdrive/lib-uhs2  -I./src/sd/usb_flashdrive/lib-uhs3  -I./src/sd/usb_flashdrive/lib-uhs3/dyn_SWI  -I./src/sd/usb_flashdrive/lib-uhs3/UHS_host  -I./src/sd/usb_flashdrive/lib-uhs3/UHS_host/UHS_BULK_STORAGE  -I./src/sd/usb_flashdrive/lib-uhs3/UHS_host/USB_HOST_SHIELD  -I.  -c -o src/lcd/utility/u8g_state.o src/lcd/utility/u8g_state.c
arm-linux-gnueabihf-g++ -O0 -c     -std=c++11 -D PANDAPI  -I./lib  -I./lib/NeoPixel_LED  -I./lib/TMCStepper-master  -I./lib/TMCStepper-master/examples  -I./lib/TMCStepper-master/examples/Calibrate_spreadCycle  -I./lib/TMCStepper-master/examples/Live_tune  -I./lib/TMCStepper-master/examples/Simple  -I./lib/TMCStepper-master/examples/Software_SPI  -I./lib/TMCStepper-master/examples/StallGuard  -I./lib/TMCStepper-master/examples/StallGuard_TMC2209  -I./lib/TMCStepper-master/examples/StallGuard_TMC2660  -I./lib/TMCStepper-master/examples/TMC_AccelStepper  -I./lib/TMCStepper-master/extras  -I./lib/TMCStepper-master/src  -I./l [... cut ...]
/usr/bin/ld: G34_M422.cpp:(.text+0x52c): undefined reference to `ZStepperAlign::xy'
/usr/bin/ld: src/gcode/calibrate/G34_M422.o: in function `GcodeSuite::M422()':
G34_M422.cpp:(.text+0xbd8): undefined reference to `ZStepperAlign::reset_to_default()'
/usr/bin/ld: G34_M422.cpp:(.text+0xf60): undefined reference to `ZStepperAlign::xy'
stderr:collect2: error: ld returned 1 exit status
stderr:make: *** [Makefile:40: pi_marlin] Error 1
make: Leaving directory '/home/pi/PandaPI/Marlin2.x/pandapi'
compile failed !stdclose:2compile failed !

markniu, 2020-09-01:
Hi, thanks for report this compile error of ZStepperAlign, I have just fixed and updated it to github. you can update the source code from the compile web page.

EMDM25, 2020-09-01:
Now I have two Problem the first one is that the

1) [BLtouch not deploying when Auto Homing]
  
2) The Seco one is that I have the sliceengineering Thermistor, I changed the config file to the #67, but it does not show the temp.

EMDM25, 2020-09-01:
Sorry  the thermistor is working just  the bltouch is the one that is not working for me for homing

markniu, 2020-09-02:
> 
> 
> Now I have two Problem the first one is that the
> 
>     1. [BLtouch not deploying when Auto Homing]
> 
>     2. The Seco one is that I have the sliceengineering Thermistor, I changed the config file to the #67, but it does not show the temp.

Check if the BLTOUCH is enabled in the marlin and the connection? and send M280 to test it at first.for example, Push-pin Down(M280 P0 S10) Push-pin Up(M280 P0 S90).
the firmware in the mcu on pandapi only support thermistor of 100k( 3950,NTC..),the temperature value maybe not correct if use other thermistors.

EMDM25, 2020-09-02:
[image]

EMDM25, 2020-09-02:
that is when you turn the printer

EMDM25, 2020-09-02:
[image]

EMDM25, 2020-09-02:
but when I send command does not respond.

EMDM25, 2020-09-02:
[image]

EMDM25, 2020-09-02:
[image]

markniu, 2020-09-02:
how about send M280 P0 S10   , if the pin can be pushed pin down?

EMDM25, 2020-09-02:
I have tried to send all the command for the bltouch and it didn't work

EMDM25, 2020-09-02:
Here is my Conf file if that can help you, is possible that I did something wrong 
[Configuration.zip](https://github.com/markniu/PandaPi/files/5160119/Configuration.zip)

EMDM25, 2020-09-02:
CONFIGURATION.H

/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#define CONFIGURATION_H_VERSION 020006

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * https://reprap.org/wiki/Calibration
 * https://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * https://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * https://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * https://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// config/examples/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a SCARA printer start with the configuration files in
// config/examples/SCARA and customize for your machine.
[... 2260 lines cut ...]

/**
 * R/C SERVO support
 * Sponsored by TrinityLabs, Reworked by codexmas
 */

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// (ms) Delay  before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

// Allow servo angle to be edited and saved to EEPROM
//#define EDITABLE_SERVO_ANGLES

EMDM25, 2020-09-02:
CONFIGURATION_ADV

/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration_adv.h
 *
 * Advanced settings.
 * Only change these if you know exactly what you're doing.
 * Some of these settings can damage your printer if improperly set!
 *
 * Basic settings can be found in Configuration.h
 *
 */
#define CONFIGURATION_ADV_H_VERSION 020006

// @section temperature

//===========================================================================
//============================= Thermal Settings ============================
//===========================================================================

/**
 * Thermocouple sensors are quite sensitive to noise.  Any noise induced in
 * the sensor wires, such as by stepper motor wires run in parallel to them,
 * may result in the thermocouple sensor reporting spurious errors.  This
 * value is the number of errors which can occur in a row before the error
 * is reported.  This allows us to ignore intermittent error conditions while
 * still detecting an actual failure, which should result in a continuous
 * stream of errors from the sensor.
 *
 * Set this value to 0 to fail on the first error to occur.
 */
#define THERMOCOUPLE_MAX_ERRORS 15

//
// Custom Thermistor 1000 parameters
//
#if TEMP_SENSOR_0 == 1000
  #define HOTEND0_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND0_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND0_BETA                 3950    // Beta value
#endif

#if TEMP_SENSOR_1 == 1000
  #define HOTEND1_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND1_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND1_BETA                 3950    // Beta value
#endif
[... 3422 lines cut ...]
#if ENABLED(PRINTCOUNTER)
  #define SERVICE_WARNING_BUZZES  3
  // Activate up to 3 service interval watchdogs
  //#define SERVICE_NAME_1      "Service S"
  //#define SERVICE_INTERVAL_1  100 // print hours
  //#define SERVICE_NAME_2      "Service L"
  //#define SERVICE_INTERVAL_2  200 // print hours
  //#define SERVICE_NAME_3      "Service 3"
  //#define SERVICE_INTERVAL_3    1 // print hours
#endif

// @section develop

//
// M100 Free Memory Watcher to debug memory usage
//
//#define M100_FREE_MEMORY_WATCHER

//
// M43 - display pin status, toggle pins, watch pins, watch endstops & toggle LED, test servo probe
//
//#define PINS_DEBUGGING

// Enable Marlin dev mode which adds some special commands
//#define MARLIN_DEV_MODE

EMDM25, 2020-09-02:
PINS_PANDA_PI

#ifndef BOARD_NAME
  #define BOARD_NAME "PANDA,PI"
#endif

#define BOARD_INFO_NAME      "PANDA,PI"

//
// Servos
//

#define SERVO0_PIN       255// 0xBE

//
#define X_MIN_PIN          1 
#define X_MAX_PIN          1 

#define Y_MIN_PIN          4 
#define Y_MAX_PIN          4 

#define Z_MAX_PIN          11 
#define Z_MIN_PIN            11 

#ifndef FIL_RUNOUT_PIN
  #define FIL_RUNOUT_PIN    255
#endif

//
// Z Probe (when not Z_MIN_PIN)
//
//#ifndef Z_MIN_PROBE_PIN
//  #define Z_MIN_PROBE_PIN  7
//#endif

//
// Steppers
//
#define X_STEP_PIN         7 
#define X_DIR_PIN          2 
#define X_ENABLE_PIN       0 

#define Y_STEP_PIN         12 
#define Y_DIR_PIN          5 
#define Y_ENABLE_PIN       0 

#define Z_STEP_PIN         14 
#define Z_DIR_PIN          13 
#define Z_ENABLE_PIN       0 

#define E0_STEP_PIN        10 
#define E0_DIR_PIN         6 
#define E0_ENABLE_PIN      0 

#define E1_STEP_PIN        26
#define E1_DIR_PIN         25
#define E1_ENABLE_PIN      0

#define E2_STEP_PIN        31
#define E2_DIR_PIN         21
#define E2_ENABLE_PIN      0
/*
#define Z2_STEP_PIN        31
#define Z2_DIR_PIN         21
#define Z2_ENABLE_PIN      0
*/

/**
 * Default pins for TMC software SPI
 */

#if HAS_TRINAMIC
  #define X_HARDWARE_SERIAL  customizedSerial1
 //  #define X2_HARDWARE_SERIAL customizedSerial1
  #define Y_HARDWARE_SERIAL  customizedSerial1
 //  #define Y2_HARDWARE_SERIAL customizedSerial1
  #define Z_HARDWARE_SERIAL  customizedSerial1
  #define Z2_HARDWARE_SERIAL customizedSerial1
  #define E0_HARDWARE_SERIAL customizedSerial1
  //#define E1_HARDWARE_SERIAL customizedSerial1
  //#define E2_HARDWARE_SERIAL customizedSerial1
  //#define E3_HARDWARE_SERIAL customizedSerial1
  //#define E4_HARDWARE_SERIAL customizedSerial1

#endif

//
// Augmentation for auto-assigning RAMPS plugs
//
#if DISABLED(IS_RAMPS_EEB) && DISABLED(IS_RAMPS_EEF) && DISABLED(IS_RAMPS_EFB) && DISABLED(IS_RAMPS_EFF) && DISABLED(IS_RAMPS_SF) && !PIN_EXISTS(MOSFET_D)
  #if HOTENDS > 1
    #if TEMP_SENSOR_BED
      #define IS_RAMPS_EEB
    #else
      #define IS_RAMPS_EEF
    #endif
  #elif TEMP_SENSOR_BED
    #define IS_RAMPS_EFB
  #else
    #define IS_RAMPS_EFF
  #endif
#endif

//
// Heaters / Fans
//
#ifndef MOSFET_D_PIN
  #define MOSFET_D_PIN     -1
#endif

// heat connector index
#define HOTBED_CODE       0  
#define HOTEND_0_CODE     1  
#define HOTEND_1_CODE     2  

#define HEATER_0_PIN     255
#define HEATER_1_PIN     255
#define HEATER_BED_PIN   255
#define TEMP_0_PIN       255
#define TEMP_1_PIN       255
#define TEMP_BED_PIN     255
#define TEMP_BOARD_PIN   255

#define FAN_PIN         255
#define FAN1_PIN         255
#define FAN2_PIN         255
#define FAN3_PIN         255

//////////////////////////
// LCDs and Controllers //
//////////////////////////
#if ENABLED(ULTRA_LCD)
  // LCD Display output pins
  #ifdef REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
	#define LCD_PINS_RS      28 
	#define LCD_PINS_ENABLE  29 
	#define LCD_PINS_D4      27  
  #endif

	#define SD_DETECT_PIN   255   
   // LCD Display input pins
  #if ENABLED(NEWPANEL)
	#define BEEPER_PIN		24 
	#define BTN_ENC 		30 
	#define BTN_EN1 		22 
	#define BTN_EN2 		23 
  #endif // NEWPANEL

#endif // ULTRA_LCD

markniu, 2020-09-02:
> that is when you turn the printer

the z_min is triggered when you turn on the printer?

EMDM25, 2020-09-02:
Yes it does

markniu, 2020-09-02:
check the connection of the z_min, it seems that there is no connection for the z_min

EMDM25, 2020-09-02:
I check it a couple of time and tried the bltouch  in another printer it did work

markniu, 2020-09-02:
please plug in one jumper to short the pins of z_min and GND,and to see if the z_min is triggered?

EMDM25, 2020-09-02:
Like that 

[image]

markniu, 2020-09-02:
yes, short zmin and G

EMDM25, 2020-09-02:
it blink on little light on the bltouch

EMDM25, 2020-09-02:
I tried again the same Flashin LED

markniu, 2020-09-02:
there is some problem of the configuration.h, I just met your problem with your configure

EMDM25, 2020-09-02:
Could you tell me were if is possible?

markniu, 2020-09-02:
disable this "USE_PROBE_FOR_Z_HOMING" in the configuration.h , I am also testing it with enable and disable it

EMDM25, 2020-09-02:
I enable "USE_PROBE_FOR_Z_HOMING" when I first have the problem, I disable it right now and tried out,  same home X home Y, but Z does not deploy BLTOUCH.

markniu, 2020-09-02:
I just test with all your configure, it works well when homing.the problem maybe on the pwm signal pin on the BLTOUCH because you have saied that there is no action after sending M280 P0 S10.

markniu, 2020-09-02:
could you check again  pins of G 5V S for BLTOUCH

markniu, 2020-09-02:
[image]

EMDM25, 2020-09-02:
It did work

EMDM25, 2020-09-02:
I got confuse because the light was blinking all the time

EMDM25, 2020-09-02:
Thanks so much

markniu, 2020-09-02:
that's great it works.  if any problem please feel free to let me know.

EMDM25, 2020-09-02:
Have you seen this error before doing  G34 

 Send: G34
Recv: echo:Unknown command: "\x88ûÜ"
Recv: 
Recv: ITERATION: 1
Recv: Probing failed.
Recv: G34 aborted.
Recv: echo:Unknown command: "\x90ûÜ¾"
Recv: ok
Recv:

EMDM25, 2020-09-03:
I know now why is the error, looks that somehow if I connect the BLtouch like this image, it works just for g29, but it will be open all-time at least when I am using G34. 

[image]

EMDM25, 2020-09-03:
and if I connect the Bltouch to Zmin and G will be  TRIGGERED all the time. I enable #define Z_MIN_ENDSTOP_INVERTING true I didn't work.