markniu/PandaPi #27: [feature-request]

Moorviper, 2020-12-03:
Hi after i have seen some Raspberry Pi shields it maybe would be great if a fan mount could be added to get a better cooling of the Raspberry PI.

[image]

Something like this:
[image]

Ok maybe routing the traces isn't possible ...
and the PCB gets bigger ...

markniu, 2020-12-03:
That's good idea!it maybe possible.I am considering how to add one fan on top of the motor driver and Pi.

Moorviper, 2020-12-04:
Or to make it easier

only opening above the SOC
(i don't have checked how much the orientation changes between the different raspberries)
and 4 holes to add a 3d printed fan shroud for the fan 

[image]
[image]
[image]