markniu/PandaPi #29: G33 failing consistently

AmEv7Fam, 2021-03-27:
I was able to attach my PandaPi to my Openbeam Kossel Pro, and all that's left is configuration. However, whenever I run G33, it consistently errors out with either "homing failed" (if I don't home first) or "Z probe failed". 

Configuration.h:

```
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

// #define CONFIG_EXAMPLES_DIR "delta/kossel_pro"

/**
 * Example configuration file for OpenBeam Kossel Pro
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 */
#define CONFIGURATION_H_VERSION 020006

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * https://reprap.org/wiki/Calibration
 * https://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * https://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * https://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * https://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
[... 2677 lines cut ...]

#define CASE_LIGHT_PIN 255

//////////////////////////
// LCDs and Controllers //
//////////////////////////
#if ENABLED(ULTRA_LCD)
  // LCD Display output pins
  #ifdef REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
	#define LCD_PINS_RS      28 
	#define LCD_PINS_ENABLE  29 
	#define LCD_PINS_D4      27  
  #endif

	#define SD_DETECT_PIN   255   
   // LCD Display input pins
  #if ENABLED(NEWPANEL)
	#define BEEPER_PIN		24 
	#define BTN_ENC 		30 
	#define BTN_EN1 		22 
	#define BTN_EN2 		23 
  #endif // NEWPANEL

#endif // ULTRA_LCD
```

thinkyhead, 2022-01-20:
Maybe…
```
#define Z_PROBE_LOW_POINT -10
```

AmEv7Fam, 2022-02-27:
> Maybe…
> 
> ```
> #define Z_PROBE_LOW_POINT -10
> ```

Tried that; made no difference.

thinkyhead, 2022-03-13:
CC: @LVD-AC