markniu/PandaPi #30: [suggestion/question] Compute Module/CM4

NickSlash, 2021-03-30:
I've never used the compute module(s) and it's virtually impossible to get the latest CM4 at the minute but they seen like a good fit for this?

It would mean extra components on your board for the io you lose from the full size pi but youd also save a bit of space too and the soc moves back to the top which would make cooling easier.

No idea how complicated implementing something like this is and I guess most people have spare pi's and not compute modules so would add more expense but still.

Just a thought.

Moorviper, 2021-04-01:
In general the CM are a great idea...
But they change the socket too often so it makes no sense to use them.

A normal rpi with the pi hat is still a better solution in the long run.

burk80, 2021-05-07:
Ive been thinking of playing with this as a fun project .. if we get a few people with cm4 to test it maybe mark can double check the cad Files?