markniu/PandaPi #31: [support] pandaPI V2.5 - Sensorless homing

adepier, 2021-03-30:
I cant make sensorless homing working with Pandapi v2.5, RaspberryPi4, TMC2209... could you share your working configuration.h and Configuration_adv.h file to compare? or update the doc, giving a troubleshooting procedure? thanks here are my files...
[Configuration.zip](https://github.com/markniu/PandaPi/files/6230011/Configuration.zip)

markniu, 2021-04-01:
you can try to change value of the sensitivity  X_STALL_SENSITIVITY Y_STALL_SENSITIVITY, different motors may have different current and sensitivity.and do not forgot to unplug the endstops.

adepier, 2021-04-01:
That's a good idea, bit still doesn't work... when I send M122, I see that TMC  is disable and stealthChop is disable
any other idea?
thanks
```
Send: M122
Recv: 		X	Y	Z	E
Recv: Address		0	1	2	3
Recv: Enabled		false	false	false	false
Recv: Set current	800	600	400	800
Recv: RMS current	1436	1049	718	1436
Recv: MAX current	2025	1479	1012	2025
Recv: Run current	25/31	18/31	12/31	25/31
Recv: Hold current	12/31	9/31	6/31	12/31
Recv: CS actual	0/31	0/31	0/31	0/31
Recv: PWM scale
Recv: vsense		0=.325	0=.325	0=.325	0=.325
Recv: stealthChop	false	false	false	false
Recv: msteps		256	256	256	256
Recv: tstep		0	0	0	0
Recv: PWM thresh.	1581	1581	2635	2668
Recv: [mm/s]		100	100	3	30
Recv: OT prewarn	false	false	false	false
Recv: pwm scale sum	0	0	0	0
Recv: pwm scale auto	0	0	0	0
Recv: pwm offset auto	0	0	0	0
Recv: pwm grad auto	0	0	0	0
Recv: off time	0	0	0	0
Recv: blank time	16	16	16	16
Recv: hysteresis
Recv:  -end		-3	-3	-3	-3
Recv:  -start		1	1	1	1
Recv: Stallguard thrs	0	0	0	0
Recv: uStep count	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	E
Recv: sg_result	0	0	0	0
Recv: stst		*	*	*	*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0x00:00:00:00	 Bad response!
Recv: 		Y	0x00:00:00:00	 Bad response!
Recv: 		Z	0x00:00:00:00	 Bad response!
Recv: 		E	0x00:00:00:00	 Bad response!
Recv: 
Recv: 
Recv: Testing X connection... Error: All LOW
Recv: Testing Y connection... Error: All LOW
Recv: Testing Z connection... Error: All LOW
Recv: Testing E connection... Error: All LOW
Recv: ok
Recv: 
```

markniu, 2021-04-03:
The reason of this TMC Error message is that with write only access to TMC, and there is no RXD connection in the hardware. So that's normal.https://github.com/markniu/PandaPi/wiki/How-to-run-TMC2209-with-UART-mode-(V2.0-2.5)
but still no idea why it does not work in your side!