markniu/PandaPi #32: OctoPrint - Python 3?

jneilliii, 2021-04-01:
Is the current flashable img files for PandaPi using OctoPrint in python 2 or python 3 venv?

Moorviper, 2021-04-01:
i am currently downloading the latest image and check

Moorviper, 2021-04-01:
[image]
nah still python2

Moorviper, 2021-04-04:
I built an octopi 0.18 image with the old 4.x rt kernel.
I had problems with haproxy.
I built the 5.4 kernel manually and will test it tomorrow.

Moorviper, 2021-04-05:
[image]

It boots. Next stepp marlin ...

Moorviper, 2021-04-07:
ready to use octopi 0.18 image with 5.10.25
http://octopi.pandapi.org/octopi0.18rt.img.zip
plain 0.18 image with kernel, overlays, modules working.

markniu, 2021-04-07:
great work!  does this work for both PI3 and PI4 ?

Moorviper, 2021-04-07:
i can't test it. I only have 2 raspberry pi 3+
I never needed a pi4.