markniu/PandaPi #33: Wired network support on Pi 3B+?

larskotthoff, 2021-04-01:
I can't get the wired networking interface to work with the latest PandaPi 2.5 image on a Raspberry Pi 3B+. The interface does not show up, nor is there anything in the logs indicating that is has been recognised. `lshw` shows the position on the USB bus as "unassigned".

Is support for the wired ethernet interface included in the kernel?

Moorviper, 2021-04-01:
I flashed the PandaPi3_20210320.img image today and it worked fine on a 3b+
i only had to extract it with keka instead of the normal mac os extractor.
the only downside was that you coudn't update octoprint anymore because the image is based on octoprint 0.16 which used python2 which is EOL since the beginning of 2020.

Have you edited any config file ?
If you boot it without any edits it should work just fine.

[image]
[image]

I will try to build an octopi 0.18 image with a 5.4.x real time kernel tomorrow...

larskotthoff, 2021-04-02:
Thanks, I didn't edit any configuration files, just copied the image to the SD card (using `dd` on Linux) and booted. The wired network was working before (running octopi for a different printer) and seems to be working in the sense that I get a green light when I plug in the network cable. I just don't get anything in the system itself that indicates that the network port has been recognized.

markniu, 2021-04-03:
please try this new[ img file](https://drive.google.com/file/d/1G6Lvu28NoxyoVsndvDV1vUB7dHU1NDYO/view?usp=sharing), RT kerenl 64bit for raspberryPi3B, based on [albebert](https://github.com/albebert/pandapi/wiki/How-to-Flash-img-&-WIFI-setup) ,ethernet enabled.

larskotthoff, 2021-04-03:
Works, thank you!

closed 2021-04-03 by larskotthoff