markniu/PandaPi #35: Program porting process

YL-martin, 2021-07-26:
This is great. I have already purchased this motherboard. This firmware can take full advantage of the many software ecosystems of the Raspberry Pi. Can the author describe the process of porting the Marlin firmware to the Raspberry Pi in detail or briefly? Can better facilitate our secondary development on the above

Moorviper, 2021-07-26:
There is no port necessary. 
Marlin can run on linux machines.
https://twitter.com/MarlinFirmware/status/1388463841210667010?s=20

markniu, 2021-07-27:
There are comments "PANDAPI" in each of the code that need to change or add.

markniu, 2021-07-27:
> 
> 
> There is no port necessary.
> Marlin can run on linux machines.
> https://twitter.com/MarlinFirmware/status/1388463841210667010?s=20

 that's great! have not try that, looking forward it!