markniu/PandaPi #42: unused PWM output signal

GHubensG, 2022-05-31:
hello, i can't find a forum to ask my question so i put it here.

im thinking about implementing the PandaPi for my lasercutter because my SKR board with a 32bit proccesor can't handle the amount of gcode i'm trying to send when lasering an image in grayscale.

The laser needs a PWM signal for control. i saw in the marlin file [pins_PANDA_PI.h](https://github.com/markniu/PandaPi/blob/master/Marlin2.x/pandapi/pins_PANDA_PI.h) a Servo pin on pin 255// 0xBE (this was not writen down in the documentation)

is it possible to use this pin? or is there another way i can get an analog output? i don't want to use a fan pin because this will get limited by the STM32 processor i think.

my setup: 
2 Nema23 stepper in CoreXY setup with 4 proximity sensors for XY-min&max. i'm Controlling the laser with M3 S1-S255 and M5 S0 commands from marlin.

rndsumgen, 2022-06-07:
https://marlinfw.org/docs/configuration/2.0.9/laser_spindle.html

I was looking through the same file for the same thing for the same reasons. Noticed it missing curious for a response.  Would love to see/hear about it having the ability to work on CNC/Laser/Extruder all out of the box. https://marlinfw.org/docs/configuration/2.0.9/laser_spindle.html goes into details about setting it up but PandaPi seems different. 

I'm guessing pin 255 is probably a software defined PWM pin?  That's not normal in marlin. 

```
#define HEATER_1_PIN     255
#define HEATER_BED_PIN   255
#define TEMP_0_PIN       255
#define TEMP_1_PIN       255
#define TEMP_BED_PIN     255
#define TEMP_BOARD_PIN   255

#define FAN_PIN         255
#define FAN1_PIN         255
#define FAN2_PIN         255
#define FAN3_PIN         255

#define CASE_LIGHT_PIN 255
```
https://github.com/markniu/PandaPi/blob/master/Marlin2.x/pandapi/pins_PANDA_PI.h

markniu, 2022-06-07:
because the Pins of heating and Fans are connect and controlled by the stm32 not raspberry pi, and we need to let marlin which runs on Pi to ignore these pins by defined them as 255.

markniu, 2022-06-07:
we can use microUSB to send gcode to control Rpi-marlin and do not forget to plugin one jumper on the P2 which beside the microUSB 
[image]

markniu, 2022-06-07:
do not know if we can set the gpio of raspberry pi as PWM, but I think the use the gpio on stm32  as PWM pin should be better .

GHubensG, 2022-06-08:
> do not know if we can set the gpio of raspberry pi as PWM, but I think use the gpio on stm32 as PWM pin should be better .

But then the amount of gcodes that is send to control my laser is limited by de STM32, that is why i wanted to use the RPI..