markniu/PandaPi #45: Z motor PINs for Klipper

patofoto, 2023-01-10:
Not able to make my Z motor working properly. Had been running Marlin with OctoPrint with our problems until I decided to change to Klipper. Have everything wired properly and have checked that all the cables are also working. Both X and Y are working properly also. Only Z is not working. This are my settings on my printer.cfg file:

```
[stepper_z]
step_pin: PB3  #check pin number. motor not moving
dir_pin: !PA15
enable_pin: !PC12 #check. suspect not the problem. energizing properly when issuing a STEPPER_BUZZ STEPPER=stepper_z
microsteps: 16
rotation_distance: 40
endstop_pin: ^!PA1 #check pin number. not triggering
position_endstop: 0.0
position_max: 5
```

Motor not moving at all or Z end stop triggers. I suspect PIN number is wrong for step_pin and endstop_pin.
I have looked over PIN designation on the Pins Map page. They match my current setup. What can be wrong?. Any help would be greatly appreciated.

markniu, 2023-01-10:
did the Z motor locked? if it can be locked or unlock that means the enable_pin works

patofoto, 2023-01-10:
Thank you for your time. Yes. The motor locks. enable_pin is correct. Can the numbers be wrong for the step_pin and the end stop_pin?.

markniu, 2023-01-10:
yes your pin numbers are right. which image of the OS are you using? or you can use other unused driver socket like E1 for the Z stepper.

patofoto, 2023-01-10:
Oh, I see a possible mistake I made. I am using a regular image of Mainsail OS. Also, even if I use another socket, the stop_pin for Z is also not working properly. It is not triggering at all.

markniu, 2023-01-10:
the reason should be the gpios on the raspberrypi influence the gpios from the stm32, for fixed that there is a python code after raspberrypi boot  in the pandapi image that will let all the gpios of raspberrpi as high input.

patofoto, 2023-01-10:
Understood. Thank you. Will try doing it correctly now.

patofoto, 2023-01-10:
Following the Klipper install steps here:

[https://github.com/markniu/PandaPi/wiki/How-to-install-klipper]

When running install-octopi.sh on my newly flashed image I get this error:

```
Ign:18 http://deb.debian.org/debian buster/main arm64 virtualenv all 15.1.0+ds-2
Err:18 http://deb.debian.org/debian buster/main arm64 virtualenv all 15.1.0+ds-2
  404  Not Found [IP: 146.75.30.132 80]
Fetched 194 MB in 20s (9,708 kB/s)
E: Failed to fetch http://deb.debian.org/debian/pool/main/p/python-virtualenv/virtualenv_15.1.0+ds-2_all.deb  404  Not Found [IP: 146.75.30.132 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

Important?

markniu, 2023-01-11:
I recommend you use the latest regular image OS if you want work with klipper.
to fix the z motor problem you can copy the file (/home/pi/gpio_pullup.py) from the pandapi image 
and add the autorun command(python /home/pi/gpio_pullup.py) in the start shell /etc/rc.local
reference this link
[regular image](https://github.com/markniu/PandaPi/wiki/How-to-install-pandapi-to-the-latest-octopi-image#6add-the-autorun-command-in-the-start-shell-etcrclocal)