markniu/PandaPi #49: Is this project abandoned?

adaliszk, 2024-05-11:
It looks like many of the embedded images are in disarray, the drivers are outdated, the maintainer does not seem to be all that active in the issues, and the repo hasn't been updated for two years now.

Are there plans to maintain this project? Especially when we do support it with money?

markniu, 2024-05-15:
Thank you so much for the support at early age, and that support this project from 2019 to 2022, and actually there is a only one person for all the things including develop,factory make,sale ,shipping , support.  and the total number of sell is only about 310 pcs from 2019.9 to 2022.3, this project have been stopped maintain although I didn't want to, and set the board out of stock on the store page in the past 2 year.  I am very sorry and sad for that, but hope you can understand.

adaliszk, 2024-05-15:
Please add a statement in the README that it is no longer maintained and archive the repo. It is all good, the issue was really with the missing information 😉